mne-tools/mne-cpp #277: BUG: Cannot compile without charts

Eric89GXL, 2017-03-22:
Following steps to reproduce:

1. Install Qt (5.8 here) without charts
2. Clone repo
3. In QtCreator, select 5.8 debug build of `mne_browse`
4. Hit "run"

The resulting error is:
```
cd disp/ && ( test -e Makefile || /home/larsoner/Qt/5.8/gcc_64/bin/qmake -o Makefile /home/larsoner/custombuilds/mne-cpp/MNE/disp/disp.pro -spec linux-g++ CONFIG+=debug CONFIG+=qml_debug ) && /usr/bin/make -f Makefile 
Project ERROR: Unknown module(s) in QT: charts
Makefile:526: recipe for target 'sub-disp-make_first-ordered' failed
make[1]: Leaving directory '/home/larsoner/custombuilds/build-mne-cpp-Desktop_Qt_5_8_0_GCC_64bit-Debug/MNE'
Makefile:45: recipe for target 'sub-MNE-make_first-ordered' failed
make[1]: *** [sub-disp-make_first-ordered] Error 3
make: *** [sub-MNE-make_first-ordered] Error 2
15:07:22: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project mne-cpp (kit: Desktop Qt 5.8.0 GCC 64bit)
The kit Desktop Qt 5.8.0 GCC 64bit has configuration issues which might be the root cause for this problem.
When executing step "Make"
```
From a brief look at the error I assume this is because charts is used within the core MNE module (MNE disp?). But maybe the "configuration issues" thing means it's possible I did something wrong with my Qt install.

I had tried compiling code from ~6 months ago and I think it worked okay, but maybe I had to do some minor tweaks then, too.

chdinh, 2017-03-22:
Thank you! Going to fix this immediately - incorporated the error with my last PR

chdinh, 2017-03-22:
Okay Fixed it - will arrive with the next PR. Thank you for reporting it.

Eric89GXL, 2017-03-23:
FYI GitHub magic allows you to do "Closes #277." as a paragraph in your PR description (or some following comment) and it both puts a reference here to the PR and also auto-closes the issue when the PR is merged.

commit 5e5816c1 (2017-03-30): [LIB-145] Last commit closes #277
closed 2017-03-30 by LorenzE