mne-tools/mne-cpp #278: BUG: Round-trip read/write scaling issue

Eric89GXL, 2017-03-22:
Once I got `mne_browse` to compile (#277), I found an I/O issue. Steps to reproduce:

1. File->Load `sample_audvis_raw.fif`
    [image]
2. File->Save `sample_audvis_2_raw.fif`
3. File->Load `sample_audvis_2_raw.fif`
    [image]

The first and second images should match, but they don't. Not sure if it's an issue with not setting `cals` properly (in Python I think we need to set them to unity when writing `float` instead of `short`?) or merely a display problem.

chdinh, 2017-03-22:
Looks like the calibrators are not set accordingly.

Eric89GXL, 2017-03-23:
@chdinh are the `cals` for BabyMEG data unity? It's possible the fix is just to set them to unity when writing, since it looks like the writing defaults to `float`.

Eric89GXL, 2017-03-24:
I think there are actually two problems. If I read a file that has been saved in `float` instead of `short` format (e.g., using MNE-Python) *and* turn off all projectors, it seems to work fine. Using a `short` file (which changes cal, range, etc.) breaks it, and having projectors on also breaks it. The latter makes sense when looking at the writing code:
```
bool FiffStream::write_raw_buffer(const MatrixXd& buf, const SparseMatrix<double>& mult)
{
    if (buf.rows() != mult.cols()) {
        printf("buffer and mult sizes do not match\n");
        return false;
    }

    SparseMatrix<double> inv_mult(mult.rows(), mult.cols());
    for (int k=0; k<inv_mult.outerSize(); ++k)
      for (SparseMatrix<double>::InnerIterator it(mult,k); it; ++it)
        inv_mult.coeffRef(it.row(),it.col()) = 1/it.value();

    MatrixXf tmp = (inv_mult*buf).cast<float>();
    this->write_float(FIFF_DATA_BUFFER,tmp.data(),tmp.rows()*tmp.cols());
    return true;
}
```
This code will work fine when `mult` is a diagonal matrix, but fail when it is not (e.g., if projectors are on).

commit 2a17b080 (2017-03-30): [LIB-45] Fixes #278 - scaling issue
commit 54a96665 (2017-03-30): [LIB-145] Closes #278. - Clean Up
closed 2017-03-30 by LorenzE