mne-tools/mne-cpp #279: Inteoperability: Can't read output data with MNE-Python

Eric89GXL, 2017-03-22:
To try to debug #278, I went to look at the data with MNE-Python but couldn't get the file to load.

Writing data with `mne_browse` or even `ex_read_write_raw` example and then loading the result in MNE-Python, I get:
```
>>> import mne
>>> raw = mne.io.read_raw_fif('test_output.fif')
...
  File "/home/larsoner/custombuilds/mne-python/mne/io/fiff/raw.py", line 162, in _read_raw_file
    info, meas = read_meas_info(fid, tree, clean_bads=True)
  File "/home/larsoner/custombuilds/mne-python/mne/io/meas_info.py", line 820, in read_meas_info
    hr['moments'] = read_tag(fid, pos).data
  File "/home/larsoner/custombuilds/mne-python/mne/io/tag.py", line 500, in read_tag
    tag.data = _read_matrix(fid, tag, shape, rlims, matrix_coding)
  File "/home/larsoner/custombuilds/mne-python/mne/io/tag.py", line 245, in _read_matrix
    fid.seek(-(ndim + 1) * 4, 1)
IOError: [Errno 22] Invalid argument
```
`ndim` is something absurdly large and other entries are bogus, so there is something wrong either at the read end, or the write end. I'll try to investigate at the Python end further.

chdinh, 2017-03-22:
That is strange since we write the data the same way with the BabyMEG and have no problem open them with MNE-C, MNE-MATLAB and MNE-CPP. Anyhow seems that there is a matrix coding issue.

chdinh, 2017-03-22:
When reading ndim, it might be that there is a mix up of little and big endianness?

Eric89GXL, 2017-03-23:
> When reading ndim, it might be that there is a mix up of little and big endianness?

A lot of the other fields that were already filled in were nonsensical, so something earlier in the process (a pointer/offset?) is probably broken. Maybe the endianness is wrong there, but it would be surprising that other readers could handle it.

> we write the data the same way with the BabyMEG and have no problem open them ...

It looks like the BabyMEG data only accesses a limited subset of available fields (see also #280). It's possible that there are some FIF entries (even in just the old sample data FIF file) that have not been extensively tested, unless you people have been doing lots of IO with non-BabyMEG data (e.g., VectorView system data)...?

Eric89GXL, 2017-03-23:
> That is strange since we write the data the same way with the BabyMEG and have no problem open them with MNE-C, MNE-MATLAB and MNE-CPP.

Actually I think it might be because these programs all ignore relevant fields in FIF files, but somehow it still tries to write them (like in #280). Debugging in QtCreator, I don't see any entries in `m_pFiffInfo` for `FIFFB_HPI_MEAS` or `FIFFB_HPI_RESULTS`, yet the output file has them, and they appear to be junk.

Check out a screengrab of the diff between the `mne show_fiff` (in Python) of two files. `mne show_fiff` gives more complete information than the C or MATLAB utils by actually trying to read the tags instead of just displaying their identification / kinds (original file left, MNE-CPP output on the right):

[image]

Some entries on the right (MNE-CPP output) are correct (albeit sometimes reordered), like the sample rate, lowpass/highpass, etc. But entries like `FIFFB_HPI_MEAS`, which aren't represented in `m_pFiffInfo` but do somehow get written out, are corrupted.

This can explain why, in Python, `mne.io.read_raw_fif` fails, but the more tolerant `mne show_info` will read it: the former is designed to throw an error when data is corrupted, whereas the latter merely notes it as `>>> BAD`. This can also explain why MNE-C, MNE-MATLAB, etc. *can* sucessfully "read it": they don't ever fully parse the file, ignoring entries like FIFFB_HPI_MEAS altogether, and thus miss the errors.

Possible solutions:

1. Fully parse the measurement info (#280). That took a decent amount of effort to get right even in Python, though, so presumably in C++ it will be worse (but maybe it's abstracted enough it won't be, given the Python structure as a guide?).

2. At one point in MNE-Python, before we parsed everything, we simply kept a byte-for-byte copy of any `meas_info` blocks that weren't parsed, and dumped those into the output file during saving. I'm not sure if something similar could be done here (or if it was even a 100% legitimate thing to do!), and I don't have enough time to ramp myself up on the MNE-CPP architecture to tackle it (or #280) myself.

3. Alternatively, you could skip writing any blocks that are not parsed. But this is likely to break any downstream processes, including e.g. movement compensation / Maxwell filtering (or anything else that uses HPI information). So it limits the use of MNE-CPP file outputs quite a bit.

As it stands now, though, if I am right about what's going on here, then file writing is more or less unusable for any Elekta system (old or new), since those all have HPI entries IIRC. But maybe there is some other explanation that I've missed.

chdinh, 2017-03-23:
Hi Eric,

thank you for having such a deep look into it and I think you are correct with you analysis. Right now for IO we still use the reduced info which can be found here https://github.com/mne-tools/mne-cpp/blob/master/MNE/fiff/fiff_info.h

By coincident we recently ported the complete C info routines, e.g.,  https://github.com/mne-tools/mne-cpp/blob/master/MNE/mne/c/mne_raw_info.h and https://github.com/mne-tools/mne-cpp/blob/master/MNE/fiff/c/fiff_types_mne-c.h which are supposed to replace in the near future the reduced info set to have the full MNE-C functionality in place. (BTW: The mne_dipole_fit which is a 100% percent port of MNE-C dipole fit + a new gui already uses those functions.)
Before we throw out the reduced info, I still need to do some major refactoring on the old C code, i.e., structure the classes a bit more accordingly and replace most of the code with highly optimized Matrix Classes from Eigen. (Basically get rid of all ungarded pointers to prevent memory leaks which are obviously present: https://scan.coverity.com/projects/mne-tools-mne-cpp )

I think to address the mne_browse issue we should start with the info classes right away.

Anyhow, it would be great, if you could point me to the respective python code.

Eric89GXL, 2017-03-23:
Check out [read_meas_info](https://github.com/mne-tools/mne-python/blob/master/mne/io/meas_info.py#L606), it should have everything that we read into `raw.info`.

PR #283 (2017-03-28): MRG: Dont convert endianness
Closes #279.

Ready for review/merge from my end.
closed 2017-03-28 by LorenzE