mne-tools/mne-cpp #284: VERSION

DimitriPapadopoulos, 2017-03-29:
File VERSION in **`MNE-Nightly-Linux-x86_64.tar.gz`** downloaded from page [MNE Software Download](http://www.nmr.mgh.harvard.edu/martinos/userInfo/data/MNE_register/)) looks like this:
```

====  MNE Software Package ====

Version 2.7.4
SVN revision 3514
Build date is Tue Apr 26 04:11:10 EDT 2016

Copyright © 2001 - 2010 Matti Hamalainen
Athinoula A. Martinos Center for Biomedical Imaging
Massachusetts General Hospital
Charlestown, Massachusetts, USA

```

Is this the latest available version of MNE-C? Is MNE-C still developed internally using a Subversion repository? How is it related to the [mne-tools/mne-cpp](https://github.com/mne-tools/mne-cpp) repository on GitHub? I haven't been following MNE lately and I'm obviously a bit lost about what is MNE-C nowadays, where to download it from, etc.

LorenzE, 2017-03-30:
Hi Dimitri, MNE-C is still updated here and then by Matti himself. The above is indeed the latest stable version of MNE-C. However, major development efforts of the MNE suite is currently focussed in MNE-CPP and MNE Python. Deciding which branch to use mostly depends on what you plan to do. May I ask what are your plans? Cheers, Lorenz

DimitriPapadopoulos, 2017-03-30:
Oh, it's just that I had installed MNE-C for our MEG team - years ago. There was an issue with MNE-C and I had to dig into which is the latest version, etc. I'm more in genetic imaging myself, I don't do MEG.

If I understand you correctly, there are 3 "main products" under the banner of MNE, MNE-HCP put apart:
* MNE or MNE-C: not under active development any more, without its own web page and without source code, but with [minimal download/installation instructions on the current MNE site](http://martinos.org/mne/stable/install_mne_c.html)
* [MNE-Python](http://martinos.org/mne), with [its own web page which is the current MNE page](http://martinos.org/mne) and [open source on GitHub](https://github.com/mne-tools/mne-python)
* [MNE-CPP](https://www.mne-cpp.org): under active development, with [its own distinct web page](https://www.mne-cpp.org) and [open source on GitHub](https://github.com/mne-tools/mne-cpp)

My only suggestion would be to clarify the situation a little bit on the MNE web page. Things that had me confused include:
* _MNE-C_/_MNE_ is referred to as _MNE-C_ in the [current MNE page](http://martinos.org/mne) and _MNE_ in the [legacy MNE download page](http://www.nmr.mgh.harvard.edu/martinos/userInfo/data/MNE_register/).
* The web site pointed to by the ["MNE tools for MEG and EEG data analysis" organization page in GitHub](https://github.com/mne-tools) is the [MNE page](http://martinos.org/mne), but this GitHub organization hosts Git repositories for projects with distinct web sites: the [MNE-Python repository itself](https://github.com/mne-tools/mne-python) which has inherited the [MNE web site](http://martinos.org/mne), and the [MNE-CPP repository](https://github.com/mne-tools/mne-cpp) which has its [own web site](http://www.mne-cpp.org). However there is not mention of _MNE-C_ in GitHub. This got me thinking _MNE-CPP_ is actually the new version of _MNE-C_ (some commands have the same name as far as I can see) but now I do understand this is not exactly the case.

LorenzE, 2017-03-30:
I feel your confusion. This is definitely something which we need to improve and work on in the future.

closed 2017-03-30 by LorenzE