mne-tools/mne-cpp #297: Qt version

DimitriPapadopoulos, 2017-05-16:
The [MNE-CPP Documentation](http://doc.mne-cpp.org/) states:
> ### Dependencies
> &gt;= Qt 5.7.1

A recent binary build seems to have been built against Qt 5.8:
```
$ ./mne_analyze
/neurospin/local/mne-cpp/bin/mne_analyze: /usr/lib/x86_64-linux-gnu/libQt5PrintSupport.so.5: version `Qt_5' not found (required by /neurospin/local/mne-cpp/bin/mne_analyze)
/neurospin/local/mne-cpp/bin/mne_analyze: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.8' not found (required by /neurospin/local/mne-cpp/bin/mne_analyze)
...
```

Why not build MNE-CPP against Qt 5.7.1? Does MNE-CPP actually depend on Qt 5.8?

More importantly, even the most recent version of Ubuntu LTS (16.04) ships with Qt 5.5.1. The previous version (14.04) is still maintained and ships with Qt 5.2.1. **Wouldn't it be possible to depend on one of these versions? Is there really a hard dependency on Qt 5.7.1 or 5.8?**

* Right now, deploying MNE-CPP on Ubuntu using the [Linux Binaries](https://www.mne-cpp.org/index.php/download/) is complicated: you need to build and deploy Qt as well and set LD_LIBRARY_PATH - probably for MNE-CPP only.
* MNE-CPP won't be easy to package MNE-CPP in NeuroDebian either, if it has to be bundled with its own version of Qt.

LorenzE, 2017-05-16:
Hi there. The main reason for the high Qt dependency version is the internal Qt3D module. They made a lot of high level changes until Qt 5.7.1., which lead to non-compatible code and compilation errors. Hence, mne-cpp will not compile with < Qt 5.7.1 due to its dependency on Qt3D. They are still improving a lot of stuff in Qt3D (performance, memory handling and new features). Qt3D just saw too many improvements from 5.7.1 till now. That is why we are also planning to change the min dependency Qt version of mne-cpp to Qt 5.9 once it is released. Qt 5.9 will also have LTS. Regarding deploying Qt under Linux: Have you tried using the pre-build Qt binaries? The Qt maintenance/download tool is a great way to do so.

DimitriPapadopoulos, 2017-05-17:
This is too involved when deploy via NFS to >300 Linux machines. It would mean deploying Qt via NFS too, with multiple problems:

- set `LD_LIBRARY_PATH` (to tell the truth we need to do with MNE-C as well as far as I can recall),
- have users run an initialization script to set `LD_LIBRARY_PATH` and possibly other variables (as is the case with MNE-C),
- adverse interaction between the deployed Qt and the rest of the Linux system.

Eric89GXL, 2017-06-10:
You could try using `linuxdeployqt` or a similar tool, and/or rely on RPATH to be set properly (and put the necessary Qt libs in `../lib/`), but I agree Qt deployment on Linux is a pain.

er06645810, 2017-08-30:
I've been struggling to deploy on Linux recently too. I've tried to replicate the steps in the jenkins build scripts on CentOS, but haven't had much luck yet. A short guide on the wiki or in the README about how to setup for development on Linux would a nice addition.

LorenzE, 2017-08-30:
Hi Erik, did you have problems compiling Qt or mne-cpp on CentOS? Best, Lorenz

er06645810, 2017-08-30:
I was able to get Qt working after some experimentation. The installer from Qt placed the library under my user's home directory and the linker couldn't find them at first. As Dimitri pointed out, the solution in that case was to set `LD_LIBRARY_PATH` to point to the Qt lib files.

Where I stand now is that the linker can't find `libMNE1GeometryInfo.so.1` and `libMNE1Interpolation.so.1`. Both libraries are compiled and exist at the expected location in mne-cpp/lib, but they aren't being recognized for some reason. If I get it figured out I'll let you know.

I'm going to be documenting all the required settings for my own purposes, but I'd be happy to contribute a readme file or something once I have it up and running.

LorenzE, 2017-08-31:
@er06645810 That would be great. Thanks. @1DIce Could you have a look at the two libraries and their Linux compilation please? @chdinh reported the same issues/problem a week ago.

LorenzE, 2017-08-31:
@er06645810  Did you add the mne-cpp lib folder to your LD_LIBRARY_PATH folder?

1DIce, 2017-08-31:
@er06645810 That is maybe a dumb question, but did you run qmake after your last pull?

er06645810, 2017-08-31:
@LorenzE @1DIce 
I've got it worked out now. The issue was that the .pro file for MNE Analyze and ex_histogram were both missing `libMNE1Interpolation` and `libMNE1GeometryInfo`. Adding both the debug and release versions of each to the `LIBS` section fixed the problem.

It seems odd to me that it would compile on other OS's without those, but it builds without them on Mac Windows, and presumably Ubuntu. I suspect that it is related to the compiler. CentOS is a derivative of RHEL and has an old version of gcc (4.8.5 on centOS 7). 

I'm going to put together a pull request with the changes I made, since it won't adversely affect systems the build properly already, but will help with oddball Linux environments.

LorenzE, 2017-09-01:
@er06645810 Great news! Thanks for looking into this. Looking forward to your PR.

closed 2019-05-20 by chdinh

LorenzE, 2019-05-20:
As proposed by @larsoner, we added QMAKE_RPATHDIR where necessary and will update our CI scripts to use linuxdeployqt in an upcoming PR. Please let me know if you want this to be reopened.

PR #375 (2019-05-20): Add missing RPATH
Add QMAKE_RPATHDIR to MNE Scan plugins and MNE Analyze extensions as well as testframes. This will remove the necessity to set LD_LIBRARY_PATH to point to the mne-cpp lib folder on Linux. Avoiding LD_LIBRARY_PATH will make sure that no other application are linking against mne-cpp shipped qt libraries. The travis linux test script was changed to not use LD_LIBRARY_PATH. This PR fixes #297.