mne-tools/mne-cpp #300: ENH?: Control MNE lib verbosity

Eric89GXL, 2017-06-10:
The MNE library is great, but outputs a lot of messages, such as reading status etc. Could a mechanism be added to control this?

The simplest would probably be to add some `ifdef` around all of the `fprintf` calls, but that's pretty invasive from a code standpoint.

Is there some better way?

er06645810, 2017-08-30:
I think this is a great idea. I've missed some really important messages because they were burried in heaps of info log messages.

MNE Python uses function decorators to build a verbose argument into most of its methods, but I don't think that's a good solution for MNE-CPP because decorators are difficult to implement in C++, and because it would require significant manual changes to existing code.

What about building a logger class that holds a static enum member that can be set to `info`, `debug`, `warning`, `error`, or `none`? Logging could be performed by calling static methods on the logger such as `Logger.Info("Message here");` or `Logger.warn("Message here");`, but an internal if statements would ensure only messages at or above the set level would make it to the console.

A simple scripts could be written to add an `#import "logger.h"` to existing header files, and a regex script could swap out `prinf()`, `cout << `, and `qDebug()` statements for `Logger.info()`. It wouldn't be perfect because some messages that should really be warnings or errors will wind up as info, but it would at least give the user control over verbosity.

Making it thread safe would help with #282 as well.

If the core team thinks this would be useful, I'd be interested in taking a crack at it.

er06645810, 2017-08-30:
This SO answer also might point in the direction too.
https://stackoverflow.com/questions/3886105/how-to-print-to-console-when-using-qt

qDebug messages can be removed at compile time, which could be useful in a performance sensitive application like MNE Scan. Meanwhile, qCritical and qError get left as are. The issue of threadsafety still remains though.