mne-tools/mne-cpp #334: Unitialized sparse compensator in RealTimeEvokedSetModel

er06645810, 2018-05-07:
I pulled the latest commit on master today and started getting a crash (seg fault) when using the averaging plugin.

To replicate:
Qt: 5.10
OS: Ubuntu 17
Compiler: GCC 7.2

Fresh build on latest commit, download a fresh copy of the sample data set, put it in the bin folder. Start `mne_rt_server`. Start `mne_scan`. Select the FiffSimulator. 

It will crash the first time after a new build of `mne_rt_server`, but seems to be fine after that first time. That is a separate bug that I don't *think* is related to this one.

Setup the FiffSimulator. Setup the averaging plugin. Connect them, hit the start button. All is well until you switch the display to the averaging plugin. That will cause the crash.

The issues comes around [here](https://github.com/mne-tools/mne-cpp/blob/65d5aa05d2b85211360b2222224d26cc8fb20614/applications/mne_scan/libs/scDisp/helpers/realtimeevokedsetmodel.cpp#L523-L543), on the last line of `updateProjection`. The `m_matSparseCompMult` variable is not initialized, ans still has dimensions (0, 0), causing the multiplication to throw a bad dimensions exception.
```c++
        //
        // Make proj sparse
        //
        tripletList.clear();
        tripletList.reserve(m_matProj.rows()*m_matProj.cols());
        for(i = 0; i < m_matProj.rows(); ++i) {
            for(k = 0; k < m_matProj.cols(); ++k) {
                if(m_matProj(i,k) != 0) {
                    tripletList.push_back(T(i, k, m_matProj(i,k)));
                }
            }
        }

        m_matSparseProjMult = SparseMatrix<double>(m_matProj.rows(),m_matProj.cols());
        if(tripletList.size() > 0) {
            m_matSparseProjMult.setFromTriplets(tripletList.begin(), tripletList.end());
        }

        //Create full multiplication matrix
        m_matSparseProjCompMult = m_matSparseProjMult * m_matSparseCompMult;
    }
```

I looked through the git history for recent changes that might have introduced this bug, but I can't find anything suspicious. I'm considering adding some logic to initialize the the compensator matrix to an identity matrix of the right size, but I feel like that wouldn't address the underlying problem. 

Where is `m_matSparceCompMult` supposed to get initialized? The only place I see it getting set is in `updateCompensator`, but if that function is called before `updateProjection` then it looks like a similar problem will occur because `m_matSparseProjMult` will not be set yet.

Any thoughts on why this might be happening? And is anybody else able to reproduce this, or is this somehow related to my system?

er06645810, 2018-05-10:
I've confirmed that this happens on Windows as well as on a different Linux box as well.

er06645810, 2018-05-11:
I've also confirmed that `m_matSparseProjCompMult` never gets set by dropping a breakpoint and debug statement inside an if statement that evaluates to true after `m_matSparseCompMult` is set.

```c++
if (m_matSparseCompMult.size() > 0) {
   qDebug() << "This message never gets displayed."
   qDebug() << "Wherever the variable is supposed to be set, it's never getting set."
   m_matSparseProjCompMult = m_matSparseProjMult * m_matSparseCompMult;
}
```

LorenzE, 2018-05-11:
Hey there. Sorry for getting back late on this. I will take a look today and let you know about my findings.

LorenzE, 2018-05-11:
@er06645810 Do you use the m_pFiffInfo->comps field in your data? I think the error comes from not initializing the compensators properly if there are no compensators set in the incoming fiff info. Adding an init call in RealTimeEvokedSetModel::setRTESet(QSharedPointer<RealTimeEvokedSet> &pRTESet) might fix it:

    beginResetModel();
    m_pRTESet = pRTESet;

    //Generate bad channel index list
    RowVectorXi sel;// = RowVectorXi(0,0);
    QStringList emptyExclude;

    if(m_pRTESet->info()->bads.size() > 0) {
        sel = FiffInfoBase::pick_channels(m_pRTESet->info()->ch_names, m_pRTESet->info()->bads, emptyExclude);
    }

    m_vecBadIdcs = sel;

    m_fSps = m_pRTESet->info()->sfreq;

    //Create the initial SSP projector
    updateProjection();    

    //Create the initial Compensator projector
    updateCompensator(0);

    //Init list of channels which are to filtered
    createFilterChannelList(m_pRTESet->info()->ch_names);

    endResetModel();

    resetSelection();

er06645810, 2018-05-13:
@LorenzE Thanks for looking into this. I'm currently just using the sample data provided with MNE-CPP, so I should have the same data as you. Were you able to reproduce the issue?

I'll give your suggestion a shot!

LorenzE, 2018-05-14:
@er06645810 Works fine at my Win10 laptop. If the problem persists let me know and I will try to reproduce on my Linux machine.

er06645810, 2018-05-14:
@LorenzE I'm glad to hear it's working well for you! I'm running into this problem on Windows 8 and Linux. I'll test it out on Mac soon too.

I wonder what the difference on our systems could be.
We've both got the latest commit checked out...same data...

Is it possible that the sample data linked [here ](http://wiki.mne-cpp.org/index.php/Get_the_sample_data_set) on the wiki is different than the data you're using?

Does it ever crash for you the first time you connect the fiff simulator to mne_rt_server after a fresh build? I wonder if that is somehow related.

LorenzE, 2018-05-15:
@er06645810 Yes, I am using the same data set as in the link. So my compensator fields are also empty, like yours. I do not get a crash using the FiffSimulator after a fresh build. I will test on a Linux soon.

LorenzE, 2018-05-18:
@er06645810 I was able to reproduce the error on Linux. I also came up with a fix, which you can find in my latest PR.

er06645810, 2018-05-20:
@LorenzE Thanks for taking the time to fix this! I like your solution much better than my own. I'm still unclear about why this bug was only present on linux though. Some small difference in the compiler settings perhaps?

I'll confirm that the problem is fixed and then close this issue.