mne-tools/mne-cpp #369: ECG simulator in MNE Scan freezes after changing the sampling frequency

puolival, 2019-05-03:
Hi,

I was testing the ECG simulator in the MNE Scan software. If I perform the following sequence of actions:

#1 Start MNE Scan: ./mne_scan
#2 Add ECG simulator sensor plugin
#3 Start the simulation
#4 Stop the simulation
#5 Decrease sampling frequency
#6 Start the simulation
#7 Stop the simulation
#8 Increase sampling frequency
#9 Start the simulation

The program always freezes. However, first increasing then decreasing the sampling frequency does not produce the same error, so I guess it is somehow related to the current/previous status.

I am running the program on Ubuntu using the most recent build available from the website.

Br,
Tuomas

LorenzE, 2019-05-09:
Hi @puolival. Thx for reporting this. After some testing I think the problem arises from: https://github.com/mne-tools/mne-cpp/blob/f7a125f467fe745baa54615b9044bc35d06b3bbf/applications/mne_scan/plugins/ecgsimulator/ecgsimulator.cpp#L200-L204

Commenting the if clause solved the problem for me. In more detail, the `isRunning` method refers to the caller thread not to the thread in which the run method is living. This can cause the plugin to get stuck in the `wait` method because the caller thread is running until the plugin gets deleted from the stage. You are using the pre-built binaries from our website, correct?

puolival, 2019-05-11:
Yes, I used the pre-built binaries from the website.

LorenzE, 2019-05-13:
Do you want to setup the development version of mne-cpp and start a PR? There are also some other code sections where this bug needs to be addressed. I can help you setup mne-cpp, point you to the problematic code sections and guide you through the process along the way.

puolival, 2019-05-15:
Yes, I can try to do that! Should I follow these guidelines http://wiki.mne-cpp.org/index.php/Portal:Contribute?

LorenzE, 2019-05-15:
Great! Yes, pls use these guidelines.

LorenzE, 2020-05-27:
Closing this issue. The ECGSimulator has been removed from MNE Scan.

closed 2020-05-27 by LorenzE