mne-tools/mne-cpp #418: Organization wide Code of Conduct

sappelhoff, 2019-12-07:
Hi MNE-CPP folks!

We wanted to add a Code of Conduct at the mne-bids repo https://github.com/mne-tools/mne-bids/issues/315 and the discussion emerged whether we could have an organization wide code of conduct.

This is supported by GitHub, see here: https://help.github.com/en/github/building-a-strong-community/creating-a-default-community-health-file-for-your-organization

@larsoner and @agramfort are on board. What are your opinions @LorenzE et al.?

> You can add default community health files for your organization to the root of a public repository called .github that is owned by the organization. GitHub will use and display default files for any public repository in the organization that does not have its own file of that type in any of the following places ...

LorenzE, 2019-12-09:
Sounds good to me!

sappelhoff, 2019-12-09:
Cool, thanks for the swift reply. :-)

closed 2019-12-09 by sappelhoff