mne-tools/mne-cpp #458: :-1: error: LNK1104: Datei "MNE1Connectivity.lib" kann nicht geöffnet werden.

evko6688, 2020-02-04:
Since today, compiling stops with the error

:-1: error: LNK1104: Datei "MNE1Connectivity.lib" kann nicht geöffnet werden.

RDoerfel, 2020-02-04:
Hi @evko6688,
this error occurs when you still have an application running, that uses this library. Did you try to close everything and build or rebuild again?

evko6688, 2020-02-05:
Hi @RDoerfel,

yes this and I also deleted all files and pulled the whole mne-cpp again, twice, same error.

LorenzE, 2020-02-05:
@evko6688 Did you remove the shadow build as well? Should be in the same folder as your repository and named something like `build-mne-cpp-Desktop_Qt_5_14_0_MSVC2017_64bit-Release`

evko6688, 2020-02-05:
Yes. I removed it aswell.

LorenzE, 2020-02-05:
can you find out from the compiler log where it stopped and the error occurred?

LorenzE, 2020-02-07:
@evko6688 Were you able to get rid of the linking error?

evko6688, 2020-02-11:
Have another problem, cannot download/re-pull the mne-cpp, would need some TeamViewer help

LorenzE, 2020-02-11:
what have you tried so far? is it a git specific problem?

evko6688, 2020-02-11:
Sync, copy the link of my "fork", Tortoise pull, Tortoise rebase, ..., reinstall Git and Tortoise, ... seems to be a general error from my side

Access denied was a often failure during the download.

LorenzE, 2020-02-11:
I sounds you are trying to setup a completely new local copy of your remote/forked repository? I would use `clone` instead of `sync`. Also, can you try to use the cmd line (Eingabeaufforderung) instead of TortoiseGit? 

With the cmd tool navigate to your local repository and follow the instructions from section `Get the Source Code` described [here](https://mne-cpp.github.io/pages/install/buildguide.html#get-the-source-code).

evko6688, 2020-02-12:
Problem solved.
Yes, I tried to setup a new local copy of forked repository. With cmd it worked properly. Also the error: LNK1104: Datei "MNE1Connectivity.lib" kann nicht geöffnet werde does not appear any more.

So problem solved.

closed 2020-02-12 by evko6688