mne-tools/mne-cpp #480: eego sports - no trigger signal

evko6688, 2020-02-12:
Triggers are not received from the Trigger adapter (connected to a PC) connected to the eego in MNE scan

The command window from MNE scan regarding the triggers shows the following error:
QTextStream: No device
QTestStream: No device
No projector specified for these data.
No projector specified for these data.
No projector specified for these data.
... 

No projector specified for these data appears if a new stimulus is presented (e.g. checkerboard changes.

johaenns, 2020-02-12:
I receive the same notifications, nevertheless the triggers are recorded in my case. Could you check, whether you can see the triggers (channel "STIM") in the recording window and whether they show up if you save the recorded data to disk?

evko6688, 2020-02-12:
You are right, the triggers are there, just needed to scroll, sorry then the problem changes to, when I add the averaging plug-in it shows no average according to the trigger

LorenzE, 2020-02-12:
What channel is shown in the QuickControl GUI window, see screenshot below?
 
[image]

johaenns, 2020-02-13:
[image]

This is what it looks like for me. The trigger detection seems to work correctly, the averaging is hard to control, since I'm not performing any measurement.

LorenzE, 2020-02-19:
Wer you able to receive the trigger channel with your setup?

evko6688, 2020-02-21:
[image]

Yes the trigger is visible.

evko6688, 2020-02-21:
How show the averaged signal look like if it works properly (I mean color etc.)?

LorenzE, 2020-02-25:
So it seems triggers are coming in. Can you check which color you selected in the Selection tab? Maybe change the background color to black in the Other tab as well.