mne-tools/mne-cpp #481: Use of new eego amplifiers without license files

evko6688, 2020-02-12:
Is it possible to use (or change the code accordingly) the new eego amplifiers (eego sports TM), because company told me they longer provide the licensce files?

LorenzE, 2020-02-12:
From what I remember the license file is not needed by our MNE Scan plugin implementation, but by the SDK itself. Since the SDK is provided by the ANT guys, I guess the newest SDK should work without the license files now? 

Have you tried to compile the plugin with the newest SDK and connect to the amplifier without the license files present in `C:\Users\Username\Documents\Eego`?

@johaenns Which SDK version do you use? Also, do you need to use the license files?

johaenns, 2020-02-13:
I'm using SDK version 1.3.19, which doesn't need a license file.

LorenzE, 2020-02-13:
@evko6688 I think I saw the same SDK version on your PC, right?

LorenzE, 2020-02-19:
Were you able to get it to run without the license files?

evko6688, 2020-02-21:
Yes it was the same SDK version. And I have no idea why, but no both new amplifiers work smoothly as well (no "nothing connected" error).

closed 2020-02-21 by evko6688