mne-tools/mne-cpp #524: Cannot Compile LSL and TMSi plugin in MacOS.

iPsych, 2020-03-14:
With below errors, LSL and TMSi plugin is not built properly in MacOS.
Is there any possibility to build tMSI driver in MacOS?

> /Users/User/Downloads/mne-cpp-master/applications/mne_scan/plugins/tmsi/tmsi.h:45: error: 'direct.h' file not found
> #include <direct.h>
>          ^~~~~~~~~~
> 
> /Users/User/Downloads/mne-cpp-master/applications/mne_scan/plugins/lsladapter/lsladapterproducer.h:50: error: 'lsl_cpp.h' file not found
> #include <lsl_cpp.h>
>          ^~~~~~~~~~~

LorenzE, 2020-03-14:
If I remember correctly the tmsi amplifier ships with a Windows SDK only. That is why I implemented it for windows in MNE Scan. Do you have a MacOS compatible SDK at your disposal?

Regarding LSL I will try to come up with a docu page on how to build the plugin.

iPsych, 2020-03-15:
@LorenzE 
Thanks for clarifying. As far as I know, the SDK is only support Windows.
Is there any TMSi-support MNE Scan version with installer? Or should I compile them in Windows by myself?

LorenzE, 2020-03-15:
The tmsi plugin is currently excluded from the installer version of MNE Scan. So as of right now you would have to build from source.

iPsych, 2020-03-16:
@LorenzE Thank you LorenzE : )
I am trying to connect ANT-Neuro refa8 64ch EEG system with MNE Scan.
Theoretically, it should be work with tmsi plugin, right?

Bests,
J.

closed 2020-03-26 by LorenzE