mne-tools/mne-cpp #550: Filter tools in Utils library

juangpc, 2020-04-03:
Hi, whats app with this filter tools. 
For instance, check the CosineFilter class. .. See the constructor has input parameters and then it redefines them? 

https://github.com/mne-tools/mne-cpp/blob/98f9741f82d70fd489882ad2d73f8af0eeee5678/libraries/utils/filterTools/cosinefilter.cpp#L73

LorenzE, 2020-04-03:
What exactly do you mean? Are you referring to, e.g., `highpasss` being transformed into samples?

https://github.com/mne-tools/mne-cpp/blob/98f9741f82d70fd489882ad2d73f8af0eeee5678/libraries/utils/filterTools/cosinefilter.cpp#L98

juangpc, 2020-04-04:
omg. Sorry to bother with this. 
I see now. highpass is different from highpasss ...Nevermind.
What a wonderful choice for variable names.

closed 2020-04-04 by juangpc