mne-tools/mne-cpp #552: Add path of already calcualted head movements (HPI fitting)

LorenzE, 2020-04-06:
Plot the history of the subject's head movements in 3D. This gives the study conductor a better impression about the subject's status and overall data quality. @juangpc came up with the idea to plot the path as connected lines over time.

RDoerfel, 2020-04-06:
Do you want to display this in an extra graph or in the 3D View?

LorenzE, 2020-04-06:
Inside the 3D view

RDoerfel, 2020-04-07:
Honestly I think this might get a bit messy to visualize this within the head. Also the movements are small compared to head and sensor size, at least in adaults. So you might need a big screen to actualy see more than a point. I actually think it is a nice idea to trace the movement somehow, I just wanted to share my concerns on this way ;)

LorenzE, 2020-04-07:
If you see a point I guess that would be a sign that the subject did not move a lot. As soon as you start seeing lines you would immediatley know that there was a movement. I might be mistaking. We should maybe try to come up with a quick prototype to see if this approach is feasible.

An alternative would be to always show the inital position of the subject as @imsorryk proposed during our last meeting.

juangpc, 2020-04-07:
I think that even if it gets messy, which it might, it is interesting to see. A checkbox would be enough I think, to ensure that even if it gets messy... that's what you wanted to see.

imsorryk, 2020-04-07:
I think the method that @LorenzE described would accomplish demonstrating head movement from initial position. And adding a toggle to show the initial head POS would clarify if the head has rotated.

Another option. Could you trace the movement of the HPI coils? That will give you information of how the head is moving, both translation and rotation.

RDoerfel, 2020-04-07:
@imsorryk I think tracing the coils would be even more messy. In MNE-Pthon you have a Dot and an arrow in Z direction of Head Space. So the arrow gets tilted when you rotate too. This will indicate some rotation.

imsorryk, 2020-04-07:
@RDoerfel I think that arrows will work just fine then. I think that is enough information to show users the head has either rotated or tilted. Are you thinking have a toggle to show the original head POS as a dot with arrows pointing to each axis, and show the current?

RDoerfel, 2020-04-08:
I think this might be the way to go. Showing the whole path is for sure interesting, but probably more important for offline analysis?

imsorryk, 2020-04-08:
For live view, I think the solution you proposed will work best. Maybe show a dot for the initial head origin, with an arrow on the Z axis, then trace movement to current origin with updated arrow. 

In post having the path of each HPI coil could be useful for quality analysis. I know some sites they place the HPI coils on the forehead of pediatric patients. And they get fitting errors because their scanner has trouble reading the coils.  This could be useful to show optimum placement.

chdinh, 2026-05-31:
Stale enhancement from the early backlog that was never scheduled and predates the v2.3.0 refactor. Closing as part of issue-tracker cleanup; please reopen with a fresh, current description if this is still wanted.

chdinh, 2026-05-31:
This is a genuinely nice idea (credit to @juangpc) — plotting the head-movement path over time in 3D as a quality cue. It was never scheduled and the HPI/Disp3D code has since moved on. Closing the original to keep the tracker focused; if you'd like to pursue it we'd be glad to help point at the right HPI/Disp3D hooks.