mne-tools/mne-cpp #553: mne_scan compiling error

juangpc, 2020-04-07:
This method...

https://github.com/mne-tools/mne-cpp/blob/eb75b90aa61a9ccedbfd2cfaed82ff38b330284f/applications/mne_scan/plugins/fiffsimulator/fiffsimulator.cpp#L206

...avoids successful compilation.

Why raw pointer?
Why long comment?
To return a pointer to a different thing? That seems too c-ish.

LorenzE, 2020-04-07:
> avoids successful compilation.

Can you be more specific please? Compiles fine [here](https://github.com/mne-tools/mne-cpp/actions/runs/69278823).

> Why raw pointer?

Raw pointer because the ownership of this widget is going to be held somewhere else. 

> Why long comment?

Aggree. This comment should be shortened.

> To return a pointer to a different thing? That seems too c-ish.

In this case returning the raw pointer seems dangerous. However, MNE Scan's architecture is built on the idea that the individual plugins specify everything they need themselves. This also them to be completle indepdnent units which live in their own "little world". This includes control GUIs which can be used to control the plgin during runtime by the user. In order to manage all control GUIs the plugins offer their GUIs, e.g., via the setupWidget() method. This method is then called by a central MNE Scan method, which again handles putting all the different plugin GUIs together in one place. The latter also takes care of proper memory management of the GUIs. That is why I like you introducing the `std::move` call in your current PR.

juangpc, 2020-04-07:
related to the std::move idea... I would very much like to take the credit for that idea... but in reality, I found it somewhere in StackOverflow. Sorry. 😂🤫 

Related to the pointer.

closed 2020-04-07 by juangpc