mne-tools/mne-cpp #575: Avoid building Qt statically in CI

LorenzE, 2020-04-22:
Avoid building Qt statically in CI by hosting self-built static Qt version ourselves. This way we should be able to include static mne-cpp builds to our PR checks as well.

juangpc, 2020-04-22:
Would you want to generate some kind of

LorenzE, 2020-04-22:
I was more thinking of building Qt statically (and for wasm) via a Github actions workflow for different platforms, upload the zip file as artifact and host them somehwere on a server. This way we could skip the Qt compilation from source in the devbuilds.yml workflow. This would make make budiling a static version of mne-cpp much faster and we could also include static build checks into the PR checks.

PR #578 (2020-04-27): [MAINT] Add pre-built static Qt versions to CI workflows
This PR adds support for pre-built static Qt versions generated with the new generateBinaries workflow. Please note that everytime we change our Qt version in the CI workflows we have to rerun the generateBinaries workflow again. This PR also add static build checks to the PR workflow. 

Closes #575
closed 2020-04-27 by LorenzE