mne-tools/mne-cpp #659: MNE Analyze: Annotation markers are moving after scrolling through the file

RDoerfel, 2020-07-03:
[image]
As presented in the screeenshot, the annotation markers moved after scrolling them out of the view and back. They were originally placed exactly at the events in the Stim Channel. The values in the Annotation settings manager stay the same.

LorenzE, 2020-07-03:
does this happen when filters deactivated?

RDoerfel, 2020-07-03:
I'm likely to say yes, but I will double check and let you know.

RDoerfel, 2020-07-05:
I tried again and it also happens with filter disabled. But I think that is more an issue of the scrolling then the annotations. When scrolling right and back to the left end of the file, I could observe that it starts at a different point.

PR #667 (2020-07-30): [FIX] Fix data plotting when scrolling to the left in MNE Analyze
This PR includes:

- Fix data plotting when scrolling to the left (raw and filtering). Fixed a problem where the current fiff curser of the beginning data block was substracted by the filter delay even if filtering was disabled. Fixes #659
- Remove `filterAllData` function and replace usage by `reloadAllData`, which leads to an performance improvement
- Rename `updateDisplayData` to `reloadAllData`
- Comment out some qInfo outputs
- Rename `getWindowSizeBlocks` to `getTotalBlockCount`
- Disable downsampling when plotting since i [... cut]
closed 2020-07-30 by LorenzE