mne-tools/mne-cpp #666: Example ex_disp_3D segfaults on Fedora 32

m9h, 2020-07-21:
Build mne-cpp successfully following instructions on Fedora 32 with the system Qt 5.14.2 rpms 
Download the MNE Sample dataset and unpack in mne-cpp/bin/MNE-sample-data
Test build with ../mne-cpp/bin/ex_disp_3D

[WARN] QWaylandGLContext::makeCurrent: eglError: 3009, this: 0x92718e0 

Segmentation fault (core dumped)

Following the same instructions but with a non-system Qt on Ubuntu works as expected. 

Is there any additional information I can provide to determine the problem on Fedora? I will try building with a downloaded non-system Qt version.

LorenzE, 2020-07-22:
Hey @m9h Can you provide the OpenGL version which is installed on the Fedora 32 distro?

m9h, 2020-07-23:
Sure @LorenzE,

$ glxinfo | grep "OpenGL version"
OpenGL version string: 4.6 (Compatibility Profile) Mesa 20.1.3

LorenzE, 2020-07-24:
thx @m9h I was "hoping" the problem you encounter might be due to a OpenGL version not supporting tessellation shaders. But OpenGL 4.6 should be high enough in order to handle all our graphics shaders. Can you give `ex_fs_surface` and `ex_read_fwd_disp_3D` a try?

m9h, 2020-07-24:
Both of those segfault too. I downloaded your prebuilt and its working fine BTW.

m9h, 2020-07-25:
OK. On Ubuntu 18.04 I have to use a non-system Qt because they are currently at 5.9.x, so I built with 5.12.9. I just built mne-cpp with the downloaded Qt 5.12.9 on Fedora 32 and now the examples run perfectly. I am going to do the same thing with Qt 5.14.2 and I can ask about it in a Fedora Developer group.

m9h, 2020-07-25:
Well, surprisingly, building with the qt.run download of Qt 5.14.2 on Fedora 32 has exactly the same failure 
[WARN] QWaylandGLContext::makeCurrent: eglError: 3009, this: 0x7d2aa40 

Segmentation fault (core dumped)

So apparently it is a Qt 5.14.2 issue. What version of Qt do you use for the prebuilt binaries for Linux?

m9h, 2020-07-25:
I don't know if its helpful but the warning at the start that differs from the pre-built binaries:
[mhough@fedora-ibm mne-cpp_shadow]$ ../mne-cpp/bin/ex_disp_3D 
[INFO] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[WARN] QSocketNotifier: Can only be used with threads started with QThread

and then with 5.15.0 there is a new warning before it segfaults

[WARN] QObject::connect(QOpenGLContext, Unknown): invalid nullptr parameter
[WARN] QWaylandGLContext::makeCurrent: eglError: 3009, this: 0x2312820 

Segmentation fault (core dumped)

LorenzE, 2020-07-27:
For static builds we use [5.14.2](https://github.com/mne-tools/mne-cpp/blob/dfdcbd44758936f654a8405b46d5f381c555638b/.github/workflows/release.yml#L70). For dynamic builds we also use [5.14.2](https://github.com/mne-tools/mne-cpp/blob/dfdcbd44758936f654a8405b46d5f381c555638b/.github/workflows/release.yml#L276). In both cases we use [Ubuntu 16.04](https://github.com/mne-tools/mne-cpp/blob/dfdcbd44758936f654a8405b46d5f381c555638b/.github/workflows/release.yml#L12). This is indeed a stange behavior. Can you try the following:

1. Start QtCreator
2. Build mne-cpp
3. Select the ex_disp_3D example after clicking on 1 (see 1 in screenshot)
[image]
4. Press the Play button with the little bug on it (see 2 in screenshot)
5. The example should now be starting within QtCreator and open the Stack window (see screenshot below). Can you share a screenshot of the stack window after the example crashed?
[image]

m9h, 2020-07-30:
Hi @LorenzE,

Really appreciate the help. OK, I am using Fedora's System Qt 5.14.2 and Qt Creator (qt5-devel and qt-creator) as the goal is to make RPM packages for NeuroFedora. 

[image]

Cheers, 

-Morgan

LorenzE, 2020-07-31:
Cool! 

Hmm it does jump out when checking for the OpenGL version. Which exact fedora version are you on? Could you point me to an image I can install in a VM to do some debugging? This might also be a good question for the qt-interest mailing list.

m9h, 2020-08-01:
I am using an up-to-date Fedora 32 install with default window manager/windowing (GNOME/Wayland).

https://getfedora.org/en/workstation/download/

I will post on qt-interest and Fedora-devel. Thanks again for your time.