mne-tools/mne-cpp #709: Invisible for Quick Control widget of Source Localization in MNE-Scan

BiboyLin, 2020-10-21:
[image]
Both in dynamic or Static Version of MNE-Scan, there is no Quick Control widget of Source Localization as the screenshot displayed. 
MNE-CPP version: Dynamic and Static Version(0.1.6) downloaded in 2020/10/21
OS: Windows 10

BiboyLin, 2020-10-21:
Also, it is hard to follow the homepage guidance(Learn/MNE Scan/Real-time Source Localization, https://mne-cpp.github.io/pages/learn/scan_sourceloc.html) to use this function. Did **MNE** module the page mentioned changed to **Real-time Source Localization** in the current version? I will be very appreciated if you can update the guides. Thanks a lot!

RDoerfel, 2020-10-21:
Hi @BiboyLin, in order to see the plugin, you have to add the [Forward Solution Plugin](https://mne-cpp.github.io/pages/learn/scan_forward.html) as well. The documentation should cover pipelines for source localization as well.

closed 2020-11-18 by LorenzE

LorenzE, 2020-11-19:
We updated the documentation. Let me know if it works for you know. Feel free to reopen if you run into problems.

PR #727 (2020-11-18): [DOC] Update rt source localization documentation
This PR includes:

- Update documentation for the real-time source localization pipeline. Closes  #709
- Make clustering in the MNE Scan forward solution plugin the default