mne-tools/mne-cpp #720: MNE Analyze not opening high resolution BEM

imsorryk, 2020-11-09:
Unable to load higher resolution BEM files found in the Sample data set under subjects/sample/bem/ 
I am able to load the file sample-5120-5120-5120-bem.fif which is low resolution, but none of the other files seem to load. They are also not visible in the data tab, nor in the 3D settings tab. In Co-Registration the MRI Subject drop down does not show any options except for the file mentioned to work.

RDoerfel, 2020-11-10:
So far, the bem- files need a specific ending. Can you rename your file to end with `-bem.fif `?

chdinh, 2026-05-31:
Old MNE Scan / MNE Analyze report against 2020-era UI/plugin code that has since been substantially reworked. There is no current reproducer on staging. Closing as stale; please reopen with up-to-date reproduction steps if it still occurs.