mne-tools/mne-cpp #731: [BUG] ex_disp_3D object handling/initialization on MacOSX

alexrockhill, 2020-11-23:
There is some weirdness with the first example on macosx Catalina with object initialization and removal. This could maybe be a good first PR for me if others think it shouldn't be too hard.

I am testing on Ubuntu now to see if it's a platform issue.

[image]

LorenzE, 2020-11-23:
Looks like the automatic rotation is messed up when items are hidden via the GUI tree view. I suspect the rotation values are not passed to the 3D items in the data model if they are hidden (disabled). Once you show them again they are out of sync with the rest. I think this is a cross-platform issue. Automated model rotation is handled in the following function: https://github.com/mne-tools/mne-cpp/blob/65bd079676fd8157bc325fdab12128f83516d49f/libraries/disp3D/engine/view/view3D.cpp#L399

The function uses `QPropertyAnimation` which, I assume, only updates the parameters on enabled items. It is a long shot but maybe using `QParallelAnimationGroup` might fix this behavior: https://doc.qt.io/qt-5/qparallelanimationgroup.html I will try to find some good point of entry for you to start working on this. Thx!

alexrockhill, 2020-11-23:
Hmmm, I just tried it on Ubuntu and there was the same issue...

LorenzE, 2020-11-24:
I had another look a this. Turns out this is not a problem with hidden items, but with 1) the speed of the rotation and 2) with the rotation of the object around their own local coordinate system. I managed to fix 1) and pushed a PR. 2) still needs to be fixed. Any ideas @alexrockhill? Maybe we have to translate the center of mass of each 3D object to the origin, rotate via z and transform it back?

alexrockhill, 2020-11-24:
Maybe it would be easier to move the camera/viewport instead of the objects?

closed 2021-03-22 by alexrockhill