mne-tools/mne-cpp #745: Add mne-cpp/bin/resources to .gitignore

RDoerfel, 2020-12-21:


LorenzE, 2020-12-22:
This is potentially dangerous since you have to add new files to bin/resources from time to time, which then would not appear when wanting to git add them. Do you run into problems with it being not in gitignore?

RDoerfel, 2020-12-22:
Not necessarily problems, but it recognizes if I change e.g. the FiffSimulator.cfg file. Hence I have to be extra careful when I `add ` and `commit`. Further, one has to apply changes on all branches and apply them again once you `checkout `to another branch and go back again. 

So it is not really an important issue, rather a bit annoying. I guess there are not many circumstances where one would use that file that much.

LorenzE, 2020-12-22:
Yes, this is annoying. I agree. The devs and reviewers have to be extra careful when reviewing PRs ;)

RDoerfel, 2020-12-22:
Alright, I guess we could then close the issue :D

juangpc, 2020-12-22:
Please allow me to reopen (probably momentarily). 

While understanding cmake, I find it subject to debate whether we should move the binary application back into the source folder, and away from the build folder. 
This behavior is, as far as I know, not the default one in many applications, however that by itself is not good reason for change in any case. So I would like to ask:

I like the idea of out-of-source build, for many reasons. And it allows to have several builds with the same source base. OK> But if you put the binaries back there, don't you disable the at least partially the benefit of out-of-source builds?

Idea: Could we move back the binary app and its related folders into the build folder so that we can have several versions of an application, with the same name, at the same time?

This would solve the issue presented in this issue. This issue is, at a fundamental level, caused by the fact that we're keeping "production" related information in the source folder. Shouldn't we completely separate these? I am trying to understand why was this done like it is right now. I don't know enough to understand it. Could you help? 

Thanks

LorenzE, 2021-01-04:
Excellent question @juangpc ! The copying of applications to mne-cpp/bin was already present when I joined the project in 2015. From my understanding this was done just for developer convenience and easier deployment. 

Convenience: If one wants to start an application from the explorer or file browser it is easier to find that all application (executables) are present in the same folder (mne-cpp/bin). 

Deployment: When deploying mne-cpp it is easier to just package the mne-cpp/bin folder instead of searching for applications in the out-of-source build folder and then copying all the executables to one single folder. However, since the deployment was recently moved out of the qmake build process, this would mean to just add searching/copying of applications in the deployment scripts.

Overall, I think mne-cpp/bin exisits to make things easier for the devs. It would be cleaner to move the creation of it to the deployment scripts, though.

Let me know if that makes sense to you.

juangpc, 2021-01-04:
Hi, 
Thanks. That's very good to know. I think that we could keep it as it is right now, however it is nice to know the reason and the depth of the consequences if we were to change it in the future.

LorenzE, 2021-01-11:
Closing. Please reopen if needed.

closed 2021-01-11 by LorenzE