mne-tools/mne-cpp #856: reserved identifier violation

elfring, 2021-10-30:
I would like to point out that identifiers like “[`_Tp`](https://github.com/mne-tools/mne-cpp/blob/6b5de08f13d63f31ceeff2943ea5c9b08e06f203/libraries/utils/generics/circularbuffer.h#L68 "Update candidate")” and “[`_MatrixType`](https://github.com/mne-tools/mne-cpp/blob/6b5de08f13d63f31ceeff2943ea5c9b08e06f203/include/3rdParty/eigen3/Eigen/src/LU/PartialPivLU.h#L17 "Another update candidate")” [do eventually not fit](https://www.securecoding.cert.org/confluence/display/cplusplus/DCL51-CPP.+Do+not+declare+or+define+a+reserved+identifier "Do not declare an identifier which is reserved for the compiler implementation.") to the expected naming convention of the C++ language standard.
Would you like to adjust your selection for unique names?

LorenzE, 2021-11-01:
Are these [Eigen library](https://eigen.tuxfamily.org/index.php?title=Main_Page) naming?

elfring, 2021-11-01:
:thought_balloon: I suggest to avoid that such software components [depend on undefined behaviour](https://www.securecoding.cert.org/confluence/display/c/CC.+Undefined+Behavior#CC.UndefinedBehavior-ub_106 "Avoiding undefined behaviour?").

juangpc, 2021-11-01:
Is it us, not making a good use of Eigen, or is Eigen somehow not following 100% the standard?

elfring, 2021-11-01:
:thought_balloon: Would you like to distinguish remaining open issues better for the involved software libraries?

juangpc, 2021-11-01:
Yes.

elfring, 2021-11-01:
:thought_balloon: How will the chances evolve to [delete misplaced underscores from affected identifiers](https://en.cppreference.com/w/cpp/language/identifiers#In_declarations "Description for reserved identifiers")?

juangpc, 2021-11-01:
This change will probably work. Would you want to try a pr with the change?

Juan