mne-tools/mne-cpp #857: Remove unnecessary null pointer checks

elfring, 2021-10-30:
[Extra null pointer checks are not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
- [FwdCompData::fwd_free_comp_data](https://github.com/mne-tools/mne-cpp/blob/6b5de08f13d63f31ceeff2943ea5c9b08e06f203/libraries/fwd/fwd_comp_data.cpp#L206 "FwdCompData::fwd_free_comp_data function")
- [MneCTFCompData](https://github.com/mne-tools/mne-cpp/blob/6b5de08f13d63f31ceeff2943ea5c9b08e06f203/libraries/mne/c/mne_ctf_comp_data.cpp#L127 "Destructor for MneCTFCompData")
- [PluginGui](https://github.com/mne-tools/mne-cpp/blob/6b5de08f13d63f31ceeff2943ea5c9b08e06f203/applications/mne_scan/mne_scan/plugingui.cpp#L131 "Destructor for PluginGui")

juangpc, 2021-11-01:
All these are very good catches. Thanks. Would you be willing to PR the changes?

elfring, 2021-11-01:
:thought_balloon: Would you become interested to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/readability-delete-null-pointer.html "Deletion of unnecessary null pointer checks")?

-----

How do you think about to [adjust any more source code places](https://github.com/mne-tools/mne-cpp/blob/6b5de08f13d63f31ceeff2943ea5c9b08e06f203/libraries/fwd/fwd_bem_solution.cpp#L45 "Update candidate: FREE_42()")? :thinking:

juangpc, 2021-11-01:
Yes, we use it. 
As shown in different places here https://mne-cpp.github.io/pages/development/contribute.html, we use qtcreator for development and that brings clang-tidy by default. But we have an unsurmountable amount of warnings. So I guess the ones you are mentioning are probably there... We keep addressing them frequently, although not frequently enough.

elfring, 2021-11-01:
How do you think about to improve the software situation any more by influencing evolution according to development tools for automatic source code adjustments? :thinking:

juangpc, 2021-11-01:
That's a good idea. But I think that the chances of a fix generating more harm than good are, in my opinion too high. Our testing pipeline, or our coverage, aren't in great either. So it would mean that it would be hard to catch the generated problem. 
Bear in mind that some of our libraries inherit from c libraries which implement a particular way of managing memory. This means that these components live a long way from raii.

elfring, 2021-11-01:
Search patterns can be specified for questionable source code places.
They can become more sophisticated on demand, can't they? :thinking:

chdinh, 2026-05-31:
Thanks — you're right that null checks before `delete` are redundant. Many of these were cleaned up during the v2.3.0 refactoring, and the remaining ones are low-risk. Closing this broad sweep; small, targeted PRs removing specific redundant checks are welcome if you'd like to finish them off.