mne-tools/mne-cpp #891: Influence of window size on hpi fit

RDoerfel, 2022-02-27:
It seems like changing the window size used for the hpi fit during runtime has a negative effect on the fitting result. So far, I could not reproduce this using the FiffSimulator.

RDoerfel, 2022-03-05:
So, I used ex_hpiFit to run a couple of experiments to investigate the influence of window size on HPI fitting quality.

## Experiment
### Data
- 19 phantom data sets with hpi
- sampling frequencies in Hz: 600, 1000, 2000 ,2053, 3000, 5000
- hpi frequency sets in Hz: 154-166, 293-321, 586-642

### Changes Variables
- Window Size in samples: 200, 400, 600, 800, 1000, 2000, 3000

### Measures
- mean localization error in mm
- mean Goodness of Fit in mm

### Statistics
- ANOVA 

## Results
[image]
Mean localization error error for all 19 files per window size.
[image]
Mean Goodness of Fit for all 19 files per window size.

### ANOVA 
- H0:  mean error is different between window sizes: p = 0.0019960434118255765
- H0: mean gof is different between window sizes: p = 0.011718108853769449

## Summary
- would not take too much from the statistics, but I think it clearly indicates that there is little influence of increasing window size on mean localization error. Further, there is an indication that increasing window size increases the GoF, which also makes sense since noise gets averaged out. I guess this dependency on window size is more severe in noisy environments.
- Definitely does not go along with our observation of really bad fits once window size was changed in the HPI-plugin. 
- Could not reproduce the issue using the fiff-simulator
- Indicates that this problem is caused within the HPI plugin in connection with the FT Buffer.

chdinh, 2026-05-31:
Old, unreproduced report about HPI-fit window size; the HPI path has since changed. Closing as stale; please reopen with a current reproducer if observed on staging.