mne-tools/mne-cpp #905: Introduce tool for consistent formatting

LorenzE, 2022-12-06:
I think it would be worth taking a look at a tool guaranteeing consistent formatting such as: https://clang.llvm.org/docs/ClangFormat.html. You can integrate it into QtCreator and VSStudio, see https://doc.qt.io/qtcreator/creator-beautifier.html.

juangpc, 2022-12-06:
I agree.
Now the rules... my man. We need to agree on which rules make the code beautiful.

😄 just joking. I'm sure we'll agree pretty easily. 😮

LorenzE, 2022-12-06:
I do not really mind about the rules, just that it is consistent and therefore easier to read :)

LorenzE, 2022-12-09:
@juangpc I gave clang-format a try. I twas straight forward to use and install. Once installed you can do the following from your cmd line:

```
clang-format -style=LLVM -i connectivity.h // This will format the file in place
clang-format -style=LLVM connectivity.h > connectivity_formatted.h // This will create a new formatted file
clang-format -style=LLVM connectivity.h // This will output the formatted file to the cmd line
```

Integrating it to QtCreator and activate "on save formatting" was easy as well, see this guide https://doc.qt.io/qtcreator/creator-beautifier.html. Once the CMake integration is done, we will of course be able to use other IDEs as well. Most of them support clang-format.

I used the built in style LLVM, which I think would serve as a good basis for our project. Individualised styles are of course supported and can be put into a clang-format file, which could be added to the git repository.

@juangpc @gabrielbmotta What do you think?

juangpc, 2022-12-09:
the way to go. 👍🏻

LorenzE, 2022-12-13:
Formatting recursively can be achieved via (on Mac/Linux):

```
find . -iname '*.h' -o -iname '*.cpp' | xargs clang-format -style=LLVM -i
```

I started some clean up work here https://github.com/LorenzE/mne-cpp/tree/feature/formatting. Still WIP but please have a look and tell me what you think. I also started to remove some preamble stuff. Still have to make it build again and revert the changes to the eigen library.

chdinh, 2026-05-31:
A repository-wide clang-format adoption was never pursued and would churn the entire tree. Closing as stale; can be reconsidered as a dedicated initiative later.