mne-tools/mne-cpp #972: Source Localization

842300143, 2024-10-29:
Hello,

We have found a plugin named "Source Localization" in the mne_scan program, but there is no corresponding implementation in the code. 

I would like to know if there is any reference code available for source localization, 
or if it is something we need to develop ourselves.
Any pointers are welcome.

juangpc, 2024-11-04:
Hi,
Sorry. I no longer am involved with this. There are a couple of implementations of the source localization algorithm. Not sure how much is still available or running correctly. Depending on your OS and other sw, you might have to build it from source.

842300143, 2024-11-13:
Thank you for your response. 

The issue we are currently facing is that in the [official website's tutorial](https://mne-cpp.github.io/pages/documentation/scan_sourceloc.html) on source localization, we found that the source localization plugin does not seem to function, and upon examining the mnc-cpp code, we could not find the corresponding source localization code implementation.

842300143, 2024-11-13:
Currently, we have a team of about ten people who are working on modifications and contributions to the entire project. 

Could you please share the contact information for the project team? Increasing communication will help address the existing issues within the project. 

Thank you for your contributions and attention to the project.

juangpc, 2024-11-20:
I'm sorry but I don't know who, if anyone at all, is currently working on
the project. Perhaps we should talk offline. Thanks.
I'm thinking of archiving this repo.

chdinh, 2026-05-31:
Support question about the Source Localization plugin / reference code rather than a tracked defect. Closing; please use the discussions for usage help.

closed 2026-05-31 by chdinh

chdinh, 2026-05-31:
Thanks for your interest in MNE-Scan! There is indeed a working real-time source-localization path — the plugin you saw is backed by the `rtcmne` plugin (real-time clustered MNE) under `src/applications/mne_scan/plugins/rtcmne/`, which pairs with the real-time forward-solution plugin `rtfwd`. The underlying algorithms live in the `mne`/`inverse` libraries (`MinimumNorm`, `RtcMne`). Those are the entry points if you're looking for reference code to build on. For usage questions we're happy to continue in the GitHub Discussions, and contributions/improvements are very welcome.